andyrozman/RileyLinkAAPS #10: Command: Cancel Bolus

andyrozman, 2018-05-11:


andyrozman, 2018-05-17:
That puts a wrench into my plans... We could disable Extended bolus and solve problem with canceling Extended bolus that way, but suspending pump for canceling bolus might be a little much. I have to talk with other people who implemented drivers in AAPS to see if they had similar problems. We could suspend pump, then reactivate it and reapply the TBR, or just refresh local history in AAPS (and let AAPS know TBR was canceled) and let AAPS handle it from that... At the moment there is no way to disable canceling of the bolus. Thanks for letting me know...

jotomo, 2018-06-06:
Cancelling a bolus is not a frequent thing (so more an edge case that needn't be polished till perfection) but good to have when needed, so it would be nice to have it eventually.
If suspending/resuming causes no other issues (vibration, if any, shouldn't be a problem/annoyance since the user is interacting with the system), I'd go for using that to cancel the bolus. Since there's a new situation afterwards (partially delivered bolus[1]), the situation needs to be reassessed anyway, so reapplying an old temp doesn't make too much sense and a neutral temp should be a safe bet till the loop does the next iteration (and also the edge case of missed readings from this point forth should be as much as can be responsibly done). Depending on taste (and or US/EU mentality w.r.t. to safety) you could apply a zero-temp for maybe 15m for the case of missed readings.

[1] With the Combo I have multiple checks as to whether a cancellation was requested (when establishing a connection, before issuing the bolus, during bolus delivery), so when delivery has started and the (MDT) pump was suspend/resumed to cancel it, there was almost certainly a partial delivery.

jotomo, 2018-06-06:
(And don't forget to add a Temp-End record to AAPS' DB in that case. And later probably a check to detect a mismatch of a temp on the pump and AAPS DB to sync up).

dabear, 2018-06-06:
You're correct that suspending will stop the boluds delivery on medtronic pumps, but it will also stop any basal delivery. Unless you can guarantee that a resume happens is triggered right after, this could be a security risk.

andyrozman, 2018-06-06:
This action was not planned at the moment. (I intended to play with it later after everything else is done), but yes, if we would suspend pump we would resume it just afterwards, and reporting back to AAPS if action was not successful, so that user knows.

andyrozman, 2018-10-03:
I decided to read history every 5 minutes to determine status of the pump (#93) and boluses (#94). This will be partitial reading only - until last read entry is reached, so that we don't communicate too much/long with pump.

closed 2018-10-03 by andyrozman