andyrozman/RileyLinkAAPS #40: Get Basal Profile - 754 pump

alternateal, 2018-07-05:
This consistently crashes the app

07-05 17:43:10.085 8445-8577/com.gxwtech.roundtrip2 I/ShowAAPS2Activity: [Thread-3] INFO  [com.gxwtech.roundtrip2.ShowAAPS2Activity$4:386]: start Action: Get Basal Profile
07-05 17:43:10.096 8445-8577/com.gxwtech.roundtrip2 I/RileyLinkCommunicationManager: [Thread-3] INFO  [info.nightscout.androidaps.plugins.PumpCommon.hw.rileylink.RileyLinkCommunicationManager:144]: Waking pump...
07-05 17:43:10.097 8445-8577/com.gxwtech.roundtrip2 I/MedtronicUtil: [Thread-3] INFO  [info.nightscout.androidaps.plugins.PumpMedtronic.util.MedtronicUtil:192]: A7 41 20 20 8D 00 
07-05 17:43:10.198 8445-8577/com.gxwtech.roundtrip2 D/RFSpyReader: [Thread-3] DEBUG [info.nightscout.androidaps.plugins.PumpCommon.hw.rileylink.ble.RFSpyReader:66]: Got data [null] at t==9217025
07-05 17:43:13.871 8445-8463/com.gxwtech.roundtrip2 D/RFTools: [Binder:8445_1] DEBUG [info.nightscout.androidaps.plugins.PumpCommon.hw.rileylink.ble.RileyLinkBLE$1:78]: Response Count is 1C
07-05 17:43:14.648 8445-8464/com.gxwtech.roundtrip2 W/BluetoothGatt: onCharacteristicRead() - Device=00:07:80:B4:BD:30 handle=11 Status=0
07-05 17:43:14.855 8445-8577/com.gxwtech.roundtrip2 D/RFSpyReader: [Thread-3] DEBUG [info.nightscout.androidaps.plugins.PumpCommon.hw.rileylink.ble.RFSpyReader:64]: Got data [47 0B A9 6D 31 C9 5C 95 68 D5 59 56 38 D6 8E 58 F4 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 59 55] at t==9221679
07-05 17:43:14.862 8445-8577/com.gxwtech.roundtrip2 I/RFSpy: [Thread-3] INFO  [info.nightscout.androidaps.plugins.PumpCommon.hw.rileylink.ble.RFSpy:124]: writeToData: decoded radio response is A7 41 20 20 8D 09 03 37 35 34 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
07-05 17:43:14.869 8445-8577/com.gxwtech.roundtrip2 I/RileyLinkCommunicationManager: [Thread-3] INFO  [info.nightscout.androidaps.plugins.PumpCommon.hw.rileylink.RileyLinkCommunicationManager:148]: wakeup: raw response is 47 0B A9 6D 31 C9 5C 95 68 D5 59 56 38 D6 8E 58 F4 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 59 55
07-05 17:43:14.870 8445-8577/com.gxwtech.roundtrip2 I/RileyLinkCommunicationManager: [Thread-3] INFO  [info.nightscout.androidaps.plugins.PumpCommon.hw.rileylink.RileyLinkCommunicationManager:74]: Sent:A7 41 20 20 92 00
07-05 17:43:14.975 8445-8577/com.gxwtech.roundtrip2 D/RFSpyReader: [Thread-3] DEBUG [info.nightscout.androidaps.plugins.PumpCommon.hw.rileylink.ble.RFSpyReader:66]: Got data [null] at t==9221799
07-05 17:43:15.480 8445-8464/com.gxwtech.roundtrip2 D/RFTools: [Binder:8445_2] DEBUG [info.nightscout.androidaps.plugins.PumpCommon.hw.rileylink.ble.RileyLinkBLE$1:78]: Response Count is 1D
07-05 17:43:16.257 8445-8463/com.gxwtech.roundtrip2 W/BluetoothGatt: onCharacteristicRead() - Device=00:07:80:B4:BD:30 handle=11 Status=0
07-05 17:43:16.465 8445-8577/com.gxwtech.roundtrip2 D/RFSpyReader: [Thread-3] DEBUG [info.nightscout.androidaps.plugins.PumpCommon.hw.rileylink.ble.RFSpyReader:64]: Got data [47 0C A9 6D 31 C9 5C 95 67 25 71 CA 35 55 55 5C 8E 55 55 6A CA 35 55 C7 4C 8E 55 5C 6C CA 35 55 CB 45 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 C5 C5] at t==9223291
07-05 17:43:16.470 8445-8577/com.gxwtech.roundtrip2 I/RFSpy: [Thread-3] INFO  [info.nightscout.androidaps.plugins.PumpCommon.hw.rileylink.ble.RFSpy:124]: writeToData: decoded radio response is A7 41 20 20 92 01 23 00 00 2E 00 0A 23 00 14 2E 00 1C 23 00 24 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
07-05 17:43:16.493 8445-8577/com.gxwtech.roundtrip2 I/RileyLinkCommunicationManager: [Thread-3] INFO  [info.nightscout.androidaps.plugins.PumpCommon.hw.rileylink.RileyLinkCommunicationManager:87]: Received:A7 41 20 20 92 01 23 00 00 2E 00 0A 23 00 14 2E 00 1C 23 00 24 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
07-05 17:43:16.493 8445-8577/com.gxwtech.roundtrip2 D/PumpMessage: Length: 64, Original Length: 1, CommandType: GetBasalProfileSTD
07-05 17:43:16.495 8445-8577/com.gxwtech.roundtrip2 D/PumpMessage: Length: 64, Original Length: 1, CommandType: GetBasalProfileSTD
07-05 17:43:16.497 8445-8577/com.gxwtech.roundtrip2 D/MedtronicConverter: [Thread-3] DEBUG [info.nightscout.androidaps.plugins.PumpMedtronic.comm.message.MedtronicConverter:39]: Raw response before convert: 23 00 00 2E 00 0A 23 00 14 2E 00 1C 23 00 24 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
07-05 17:43:16.499 8445-8577/com.gxwtech.roundtrip2 D/MedtronicCommunicationManager: [Thread-3] DEBUG [info.nightscout.androidaps.plugins.PumpMedtronic.comm.MedtronicCommunicationManager:442]: Converted response for GetBasalProfileSTD is info.nightscout.androidaps.plugins.PumpMedtronic.comm.data.BasalProfile@ebea7ae.
07-05 17:43:16.516 8445-8445/com.gxwtech.roundtrip2 D/AndroidRuntime: Shutting down VM
07-05 17:43:16.518 8445-8445/com.gxwtech.roundtrip2 E/AndroidRuntime: FATAL EXCEPTION: main
                                                                      Process: com.gxwtech.roundtrip2, PID: 8445
                                                                      java.util.IllegalFormatConversionException: x != [B
                                                                          at java.util.Formatter$FormatSpecifier.failConversion(Formatter.java:4135)
                                                                          at java.util.Formatter$FormatSpecifier.printInteger(Formatter.java:2832)
                                                                          at java.util.Formatter$FormatSpecifier.print(Formatter.java:2786)
                                                                          at java.util.Formatter.format(Formatter.java:2491)
                                                                          at java.util.Formatter.format(Formatter.java:2426)
                                                                          at java.lang.String.format(String.java:2626)
                                                                          at info.nightscout.androidaps.plugins.PumpMedtronic.comm.data.BasalProfile.getBasalProfileAsString(BasalProfile.java:98)
                                                                          at com.gxwtech.roundtrip2.ShowAAPS2Activity.sendData(ShowAAPS2Activity.java:282)
                                                                          at com.gxwtech.roundtrip2.ShowAAPS2Activity$3.onReceive(ShowAAPS2Activity.java:143)
                                                                          at android.support.v4.content.LocalBroadcastManager.executePendingBroadcasts(LocalBroadcastManager.java:311)
                                                                          at android.support.v4.content.LocalBroadcastManager.access$000(LocalBroadcastManager.java:47)
                                                                          at android.support.v4.content.LocalBroadcastManager$1.handleMessage(LocalBroadcastManager.java:120)
                                                                          at android.os.Handler.dispatchMessage(Handler.java:102)
                                                                          at android.os.Looper.loop(Looper.java:154)
                                                                          at android.app.ActivityThread.main(ActivityThread.java:6186)
                                                                          at java.lang.reflect.Method.invoke(Native Method)
                                                                          at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:889)
                                                                          at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:779)

andyrozman, 2018-07-26:
This should work now. Change is in commit f8a78a6fdb43ab4ee5c3634f25cdc2f04c8a5d3a

alternateal, 2018-07-27:
This hasn't worked 👎 

07-27 06:33:13.158 3058-3304/com.gxwtech.roundtrip2 I/ShowAAPS2Activity: [Thread-3] INFO  [com.gxwtech.roundtrip2.ShowAAPS2Activity$4:386]: start Action: Get Basal Profile
07-27 06:33:13.158 3058-3304/com.gxwtech.roundtrip2 I/RileyLinkCommunicationManager: [Thread-3] INFO  [info.nightscout.androidaps.plugins.PumpCommon.hw.rileylink.RileyLinkCommunicationManager:116]: Waking pump...
07-27 06:33:13.172 3058-3304/com.gxwtech.roundtrip2 I/MedtronicUtil: [Thread-3] INFO  [info.nightscout.androidaps.plugins.PumpMedtronic.util.MedtronicUtil:194]: A7 41 20 20 8D 00 
07-27 06:33:13.275 3058-3304/com.gxwtech.roundtrip2 D/RFSpyReader: [Thread-3] DEBUG [info.nightscout.androidaps.plugins.PumpCommon.hw.rileylink.ble.RFSpyReader:66]: Got data [null] at t==631961
07-27 06:33:16.969 3058-3078/com.gxwtech.roundtrip2 D/RFTools: [Binder:3058_2] DEBUG [info.nightscout.androidaps.plugins.PumpCommon.hw.rileylink.ble.RileyLinkBLE$1:78]: Response Count is 78
07-27 06:33:17.939 3058-3304/com.gxwtech.roundtrip2 D/RFSpyReader: [Thread-3] DEBUG [info.nightscout.androidaps.plugins.PumpCommon.hw.rileylink.ble.RFSpyReader:64]: Got data [49 28 A9 6D 31 C9 5C 95 68 D5 59 56 38 D6 8E 58 F4 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 59 55] at t==636625
07-27 06:33:17.942 3058-3304/com.gxwtech.roundtrip2 I/RFSpy: [Thread-3] INFO  [info.nightscout.androidaps.plugins.PumpCommon.hw.rileylink.ble.RFSpy:125]: writeToData: decoded radio response is A7 41 20 20 8D 09 03 37 35 34 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
07-27 06:33:17.949 3058-3304/com.gxwtech.roundtrip2 I/RileyLinkCommunicationManager: [Thread-3] INFO  [info.nightscout.androidaps.plugins.PumpCommon.hw.rileylink.RileyLinkCommunicationManager:120]: wakeup: raw response is 49 28 A9 6D 31 C9 5C 95 68 D5 59 56 38 D6 8E 58 F4 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 59 55
07-27 06:33:17.950 3058-3304/com.gxwtech.roundtrip2 I/RileyLinkCommunicationManager: [Thread-3] INFO  [info.nightscout.androidaps.plugins.PumpCommon.hw.rileylink.RileyLinkCommunicationManager:72]: Sent:A7 41 20 20 92 00
07-27 06:33:18.055 3058-3304/com.gxwtech.roundtrip2 D/RFSpyReader: [Thread-3] DEBUG [info.nightscout.androidaps.plugins.PumpCommon.hw.rileylink.ble.RFSpyReader:66]: Got data [null] at t==636739
07-27 06:33:18.523 3058-3078/com.gxwtech.roundtrip2 D/RFTools: [Binder:3058_2] DEBUG [info.nightscout.androidaps.plugins.PumpCommon.hw.rileylink.ble.RileyLinkBLE$1:78]: Response Count is 79
07-27 06:33:19.468 3058-3304/com.gxwtech.roundtrip2 D/RFSpyReader: [Thread-3] DEBUG [info.nightscout.androidaps.plugins.PumpCommon.hw.rileylink.ble.RFSpyReader:64]: Got data [49 29 A9 6D 31 C9 5C 95 67 25 71 CA 35 55 55 5C 8E 55 55 6A CA 35 55 C7 4C 8E 55 5C 6C CA 35 55 CB 45 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 C5 C5] at t==638152
07-27 06:33:19.484 3058-3304/com.gxwtech.roundtrip2 I/RFSpy: [Thread-3] INFO  [info.nightscout.androidaps.plugins.PumpCommon.hw.rileylink.ble.RFSpy:125]: writeToData: decoded radio response is A7 41 20 20 92 01 23 00 00 2E 00 0A 23 00 14 2E 00 1C 23 00 24 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
07-27 06:33:19.491 3058-3304/com.gxwtech.roundtrip2 I/RileyLinkCommunicationManager: [Thread-3] INFO  [info.nightscout.androidaps.plugins.PumpCommon.hw.rileylink.RileyLinkCommunicationManager:88]: Received:A7 41 20 20 92 01 23 00 00 2E 00 0A 23 00 14 2E 00 1C 23 00 24 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
07-27 06:33:19.491 3058-3304/com.gxwtech.roundtrip2 D/PumpMessage: Length: 64, Original Length: 1, CommandType: GetBasalProfileSTD
07-27 06:33:19.497 3058-3304/com.gxwtech.roundtrip2 D/PumpMessage: Length: 64, Original Length: 1, CommandType: GetBasalProfileSTD
07-27 06:33:19.501 3058-3304/com.gxwtech.roundtrip2 D/MedtronicConverter: [Thread-3] DEBUG [info.nightscout.androidaps.plugins.PumpMedtronic.comm.message.MedtronicConverter:39]: Raw response before convert: 23 00 00 2E 00 0A 23 00 14 2E 00 1C 23 00 24 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
07-27 06:33:19.511 3058-3304/com.gxwtech.roundtrip2 D/MedtronicCommunicationManager: [Thread-3] DEBUG [info.nightscout.androidaps.plugins.PumpMedtronic.comm.MedtronicCommunicationManager:461]: Converted response for GetBasalProfileSTD is info.nightscout.androidaps.plugins.PumpMedtronic.comm.data.BasalProfile@6938d88.
07-27 06:33:19.529 3058-3058/com.gxwtech.roundtrip2 D/AndroidRuntime: Shutting down VM
[... cut ...]
                                                                          at java.lang.reflect.Method.invoke(Native Method)
                                                                          at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:440)
                                                                          at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:807)
07-27 06:33:19.740 3058-3076/com.gxwtech.roundtrip2 I/zygote: Background concurrent copying GC freed 22717(2MB) AllocSpace objects, 1(20KB) LOS objects, 49% free, 3MB/6MB, paused 122us total 263.116ms

andyrozman, 2018-07-30:
This works now in 554, but we still need to test it in 523.

jcwarrior, 2018-08-02:
Still fail on 523. Sorry I don't have logs because I'm running directly on phone and app crash. Is it possible to persist logcat info on a file on disk to get it when app crash?

andyrozman, 2018-08-09:
Opened #51 for reading whole profile, since it doesn';t have to do with this bug.

closed 2018-08-09 by andyrozman