andyrozman/RileyLinkAAPS #41: Get Remaining Insulin - 754 Pump

alternateal, 2018-07-05:
Returns a value of 0.0 when 139.5 units left in reservoir

07-05 17:46:03.686 8668-8750/com.gxwtech.roundtrip2 I/ShowAAPS2Activity: [Thread-3] INFO  [com.gxwtech.roundtrip2.ShowAAPS2Activity$4:386]: start Action: Status - Remaining Insulin
07-05 17:46:03.688 8668-8750/com.gxwtech.roundtrip2 I/RileyLinkCommunicationManager: [Thread-3] INFO  [info.nightscout.androidaps.plugins.PumpCommon.hw.rileylink.RileyLinkCommunicationManager:74]: Sent:A7 41 20 20 73 00
07-05 17:46:03.790 8668-8750/com.gxwtech.roundtrip2 D/RFSpyReader: [Thread-3] DEBUG [info.nightscout.androidaps.plugins.PumpCommon.hw.rileylink.ble.RFSpyReader:66]: Got data [null] at t==9390617
07-05 17:46:04.251 8668-8680/com.gxwtech.roundtrip2 D/RFTools: [Binder:8668_1] DEBUG [info.nightscout.androidaps.plugins.PumpCommon.hw.rileylink.ble.RileyLinkBLE$1:78]: Response Count is 28
07-05 17:46:04.978 8668-8681/com.gxwtech.roundtrip2 W/BluetoothGatt: onCharacteristicRead() - Device=00:07:80:B4:BD:30 handle=11 Status=0
07-05 17:46:05.117 8668-8750/com.gxwtech.roundtrip2 D/RFSpyReader: [Thread-3] DEBUG [info.nightscout.androidaps.plugins.PumpCommon.hw.rileylink.ble.RFSpyReader:64]: Got data [47 0E A9 6D 31 C9 5C 95 5A 35 74 55 55 55 C6 5B 0D 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 CA C5] at t==9391941
07-05 17:46:05.135 8668-8750/com.gxwtech.roundtrip2 I/RFSpy: [Thread-3] INFO  [info.nightscout.androidaps.plugins.PumpCommon.hw.rileylink.ble.RFSpy:124]: writeToData: decoded radio response is A7 41 20 20 73 04 00 00 15 CD 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
07-05 17:46:05.145 8668-8750/com.gxwtech.roundtrip2 I/RileyLinkCommunicationManager: [Thread-3] INFO  [info.nightscout.androidaps.plugins.PumpCommon.hw.rileylink.RileyLinkCommunicationManager:87]: Received:A7 41 20 20 73 04 00 00 15 CD 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
07-05 17:46:05.145 8668-8750/com.gxwtech.roundtrip2 D/PumpMessage: Length: 4, Original Length: 4, CommandType: GetRemainingInsulin
07-05 17:46:05.145 8668-8750/com.gxwtech.roundtrip2 D/PumpMessage: Length: 4, Original Length: 4, CommandType: GetRemainingInsulin
07-05 17:46:05.150 8668-8750/com.gxwtech.roundtrip2 D/MedtronicConverter: [Thread-3] DEBUG [info.nightscout.androidaps.plugins.PumpMedtronic.comm.message.MedtronicConverter:39]: Raw response before convert: 00 00 15 CD 
07-05 17:46:05.154 8668-8750/com.gxwtech.roundtrip2 I/System.out: Remaining insulin: 0.0
07-05 17:46:05.155 8668-8750/com.gxwtech.roundtrip2 D/MedtronicCommunicationManager: [Thread-3] DEBUG [info.nightscout.androidaps.plugins.PumpMedtronic.comm.MedtronicCommunicationManager:442]: Converted response for GetRemainingInsulin is 0.0.

andyrozman, 2018-07-22:
This should be fixed now. Please test on Veo - 554. 

Its the last commit: https://github.com/andyrozman/RileyLinkAAPS/commit/06ca38c2c428c0c55d0fef90c5c6c4248607d035

There are some other changes there and you will have to reconfigure your RL in App.

alternateal, 2018-07-26:
This can be closed on 754 pump

alternateal, 2018-07-26:
and it was with 0.9; 0.9 software

andyrozman, 2018-08-01:
Tested on 554 by ticket owner @alternateal and tested on 523 by @jcwarrior. So closing the ticket as solved.

closed 2018-08-01 by andyrozman