andyrozman/RileyLinkAAPS #49: Crash when Get Settings (Status - settings) on 523 pump

jcwarrior, 2018-08-03:
crash app on Status - settings on 523 pump

`08-03 12:25:30.076 24521-24955/com.gxwtech.roundtrip2 I/ShowAAPS2Activity: [Thread-6] INFO  [com.gxwtech.roundtrip2.ShowAAPS2Activity$4:386]: start Action: Status - Settings
08-03 12:25:30.077 24521-24955/com.gxwtech.roundtrip2 I/RileyLinkCommunicationManager: [Thread-6] INFO  [info.nightscout.androidaps.plugins.PumpCommon.hw.rileylink.RileyLinkCommunicationManager:75]: Sent:A7 93 53 69 C0 00
08-03 12:25:30.179 24521-24955/com.gxwtech.roundtrip2 D/RFSpyReader: [Thread-6] DEBUG [info.nightscout.androidaps.plugins.PumpCommon.hw.rileylink.ble.RFSpyReader:66]: Got data [null] at t==166188591
08-03 12:25:30.181 24521-24955/com.gxwtech.roundtrip2 D/RFSpy: [Thread-6] DEBUG [info.nightscout.androidaps.plugins.PumpCommon.hw.rileylink.ble.RFSpy:119]: writeToData (command=SendAndListen,raw=16 05 00 00 00 00 00 00 07 D0 00 A9 66 63 96 39 99 B1 55 55 D2 65 00 )
08-03 12:25:30.705 24521-24945/com.gxwtech.roundtrip2 D/RFTools: [Binder:24521_4] DEBUG [info.nightscout.androidaps.plugins.PumpCommon.hw.rileylink.ble.RileyLinkBLE$1:78]: Response Count is 4A
08-03 12:25:31.497 24521-24955/com.gxwtech.roundtrip2 D/RFSpyReader: [Thread-6] DEBUG [info.nightscout.androidaps.plugins.PumpCommon.hw.rileylink.ble.RFSpyReader:64]: Got data [31 14 A9 66 63 96 39 99 B1 5C 59 55 55 71 55 55 71 57 15 55 C9 55 55 C5 A5 71 55 55 55 57 15 55 55 59 B4 57 15 65 55 5C 59 55 5C 59 57 15 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 9B 45] at t==166189907
08-03 12:25:31.506 24521-24955/com.gxwtech.roundtrip2 I/RFSpy: [Thread-6] INFO  [info.nightscout.androidaps.plugins.PumpCommon.hw.rileylink.ble.RFSpy:146]: writeToData: decoded radio response is A7 93 53 69 C0 19 00 01 00 01 01 00 20 00 18 01 00 00 01 00 00 64 01 05 00 19 00 19 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
08-03 12:25:31.514 24521-24955/com.gxwtech.roundtrip2 I/RileyLinkCommunicationManager: [Thread-6] INFO  [info.nightscout.androidaps.plugins.PumpCommon.hw.rileylink.RileyLinkCommunicationManager:91]: Received:A7 93 53 69 C0 19 00 01 00 01 01 00 20 00 18 01 00 00 01 00 00 64 01 05 00 19 00 19 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
08-03 12:25:31.514 24521-24955/com.gxwtech.roundtrip2 D/PumpMessage: Length: 25, Original Length: 25, CommandType: Settings
08-03 12:25:31.517 24521-24955/com.gxwtech.roundtrip2 D/PumpMessage: Length: 25, Original Length: 25, CommandType: Settings
08-03 12:25:31.521 24521-24955/com.gxwtech.roundtrip2 D/MedtronicConverter: [Thread-6] DEBUG [info.nightscout.androidaps.plugins.PumpMedtronic.comm.MedtronicConverter:38]: Raw response before convert: 00 01 00 01 01 00 20 00 18 01 00 00 01 00 00 64 01 05 00 19 00 19 01 00 00 
08-03 12:25:31.529 24521-24955/com.gxwtech.roundtrip2 D/MedtronicCommunicationManager: [Thread-6] DEBUG [info.nightscout.androidaps.plugins.PumpMedtronic.comm.MedtronicCommunicationManager:497]: Converted response for Settings is {PCFG_CAPTURE_EVENT_ENABLE=info.nightscout.androidaps.plugins.PumpMedtronic.data.dto.PumpSettingDTO@d5b14a3, CFG_PARADIGM_LINK_ENABLE=info.nightscout.androidaps.plugins.PumpMedtronic.data.dto.PumpSettingDTO@923a3a0, PCFG_VARIABLE_BOLUS_ENABLED=info.nightscout.androidaps.plugins.PumpMedtronic.data.dto.PumpSettingDTO@dc79b59, PCFG_ALARM_MODE=info.nightscout.androidaps.plugins.PumpMedtronic.data.dto.PumpSettingDTO@f28731e, PCFG_AUDIO_BOLUS_ENABLED=info.nightscout.androidaps.plugins.PumpMedtronic.data.dto.PumpSettingDTO@e8057ff, PCFG_INSULIN_CONCENTRATION=info.nightscout.androidaps.plugins.PumpMedtronic.data.dto.PumpSettingDTO@5af59cc, PCFG_MAX_BASAL=info.nightscout.androidaps.plugins.PumpMedtronic.data.dto.PumpSettingDTO@8cd9415, CFG_MM_RF_ENABLED=info.nightscout.androidaps.plugins.PumpMedtronic.data.dto.PumpSettingDTO@fe4a32a, PCFG_INSULIN_ACTION_TYPE=info.nightscout.androidaps.plugins.PumpMedtronic.data.dto.PumpSettingDTO@a8c751b, PCFG_MM_SRESERVOIR_WARNING_POINT=info.nightscout.androidaps.plugins.PumpMedtronic.data.dto.PumpSettingDTO@75e46b8, PCFG_AUTOOFF_TIMEOUT=info.nightscout.androidaps.plugins.PumpMedtronic.data.dto.PumpSettingDTO@7005c91, PCFG_ALARM_BEEP_VOLUME=info.nightscout.androidaps.plugins.PumpMedtronic.data.dto.PumpSettingDTO@81c7f6, PCFG_BASAL_PROFILES_ENABLED=info.nightscout.androidaps.plugins.PumpMedtronic.data.dto.PumpSettingDTO@7347f7, CFG_MM_BLOCK_ENABLED=info.nightscout.androidaps.plugins.PumpMedtronic.data.dto.PumpSettingDTO@10d1664, PCFG_TEMP_BASAL_TYPE=info.nightscout.androidaps.plugins.PumpMedtronic.data.dto.PumpSettingDTO@4bd30cd, CFG_MM_KEYPAD_LOCKED=info.nightscout.androidaps.plugins.PumpMedtronic.data.dto.PumpSettingDTO@4fa6d82, PCFG_MAX_BOLUS=info.nightscout.androidaps.plugins.PumpMedtronic.data.dto.PumpSettingDTO@8a6c93, PCFG_BOLUS_SCROLL_STEP_SIZE=info.nightscout.androidaps.plugins.PumpMedtronic.data.dto.PumpSettingDTO@4b734d0, PCFG_OTHER_DEVICE_PAIRED_STATE=info.nightscout.androidaps.plugins.PumpMedtronic.data.dto.PumpSettingDTO@cf90cc9, PCFG_MM_RESERVOIR_WARNING_TYPE_TIME=info.nightscout.androidaps.plugins.PumpMedtronic.data.dto.PumpSettingDTO@465dfce, PCFG_OTHER_DEVICE_ENABLE=info.nightscout.androidaps.plugins.PumpMedtronic.data.dto.PumpSettingDTO@e53eef, CFG_BASE_CLOCK_MODE=info.nightscout.androidaps.plugins.PumpMedtronic.data.dto.PumpSettingDTO@6a2cdfc}.
08-03 12:25:31.531 24521-24521/com.gxwtech.roundtrip2 D/AndroidRuntime: Shutting down VM
08-03 12:25:31.536 24521-24521/com.gxwtech.roundtrip2 E/AndroidRuntime: FATAL EXCEPTION: main
                                                                        Process: com.gxwtech.roundtrip2, PID: 24521
                                                                        java.lang.ClassCastException: java.util.HashMap cannot be cast to java.util.List
                                                                            at com.gxwtech.roundtrip2.ShowAAPS2Activity.sendData(ShowAAPS2Activity.java:359)
                                                                            at com.gxwtech.roundtrip2.ShowAAPS2Activity$3.onReceive(ShowAAPS2Activity.java:143)
                                                                            at android.support.v4.content.LocalBroadcastManager.executePendingBroadcasts(LocalBroadcastManager.java:311)
                                                                            at android.support.v4.content.LocalBroadcastManager.access$000(LocalBroadcastManager.java:47)
                                                                            at android.support.v4.content.LocalBroadcastManager$1.handleMessage(LocalBroadcastManager.java:120)
                                                                            at android.os.Handler.dispatchMessage(Handler.java:105)
                                                                            at android.os.Looper.loop(Looper.java:164)
                                                                            at android.app.ActivityThread.main(ActivityThread.java:6944)
                                                                            at java.lang.reflect.Method.invoke(Native Method)
                                                                            at com.android.internal.os.Zygote$MethodAndArgsCaller.run(Zygote.java:327)
                                                                            at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1374)`

jcwarrior, 2018-08-03:
Works with commit 669b4c2  on 523 pump

alternateal, 2018-08-04:
Works on 754

`08-04 08:04:07.940 4268-4512/com.gxwtech.roundtrip2 I/ShowAAPS2Activity: [Thread-4] INFO  [com.gxwtech.roundtrip2.ShowAAPS2Activity$4:385]: start Action: Status - Settings
08-04 08:04:07.941 4268-4512/com.gxwtech.roundtrip2 I/RileyLinkCommunicationManager: [Thread-4] INFO  [info.nightscout.androidaps.plugins.PumpCommon.hw.rileylink.RileyLinkCommunicationManager:75]: Sent:A7 41 20 20 C0 00
08-04 08:04:08.042 4268-4512/com.gxwtech.roundtrip2 D/RFSpyReader: [Thread-4] DEBUG [info.nightscout.androidaps.plugins.PumpCommon.hw.rileylink.ble.RFSpyReader:66]: Got data [null] at t==1300758
08-04 08:04:08.043 4268-4512/com.gxwtech.roundtrip2 D/RFSpy: [Thread-4] DEBUG [info.nightscout.androidaps.plugins.PumpCommon.hw.rileylink.ble.RFSpy:119]: writeToData (command=SendAndListen,raw=16 05 00 00 00 00 00 00 07 D0 00 A9 6D 31 C9 5C 95 B1 55 55 98 B5 00 )
08-04 08:04:08.562 4268-4287/com.gxwtech.roundtrip2 D/RFTools: [Binder:4268_2] DEBUG [info.nightscout.androidaps.plugins.PumpCommon.hw.rileylink.ble.RileyLinkBLE$1:78]: Response Count is 66
08-04 08:04:09.517 4268-4512/com.gxwtech.roundtrip2 D/RFSpyReader: [Thread-4] DEBUG [info.nightscout.androidaps.plugins.PumpCommon.hw.rileylink.ble.RFSpyReader:64]: Got data [3E 1F A9 6D 31 C9 5C 95 B1 5C 59 55 55 74 57 15 65 57 15 55 58 D5 55 2F 45 71 55 55 55 57 15 55 55 59 B4 57 15 65 55 5C 74 55 58 F2 55 55 71 57 15 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 34 B5] at t==1302231
08-04 08:04:09.528 4268-4512/com.gxwtech.roundtrip2 I/RFSpy: [Thread-4] INFO  [info.nightscout.androidaps.plugins.PumpCommon.hw.rileylink.ble.RFSpy:146]: writeToData: decoded radio response is A7 41 20 20 C0 19 00 04 01 05 01 00 7D 00 B4 01 00 00 01 00 00 64 01 05 00 14 00 32 00 01 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
08-04 08:04:09.534 4268-4512/com.gxwtech.roundtrip2 I/RileyLinkCommunicationManager: [Thread-4] INFO  [info.nightscout.androidaps.plugins.PumpCommon.hw.rileylink.RileyLinkCommunicationManager:91]: Received:A7 41 20 20 C0 19 00 04 01 05 01 00 7D 00 B4 01 00 00 01 00 00 64 01 05 00 14 00 32 00 01 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
08-04 08:04:09.534 4268-4512/com.gxwtech.roundtrip2 D/PumpMessage: Length: 25, Original Length: 25, CommandType: Settings
08-04 08:04:09.536 4268-4512/com.gxwtech.roundtrip2 D/PumpMessage: Length: 25, Original Length: 25, CommandType: Settings
08-04 08:04:09.537 4268-4512/com.gxwtech.roundtrip2 D/MedtronicConverter: [Thread-4] DEBUG [info.nightscout.androidaps.plugins.PumpMedtronic.comm.MedtronicConverter:38]: Raw response before convert: 00 04 01 05 01 00 7D 00 B4 01 00 00 01 00 00 64 01 05 00 14 00 32 00 01 01 
08-04 08:04:09.546 4268-4512/com.gxwtech.roundtrip2 D/MedtronicCommunicationManager: [Thread-4] DEBUG [info.nightscout.androidaps.plugins.PumpMedtronic.comm.MedtronicCommunicationManager:497]: Converted response for Settings is {PCFG_CAPTURE_EVENT_ENABLE=key=PCFG_CAPTURE_EVENT_ENABLE,value=No,group=Other, CFG_PARADIGM_LINK_ENABLE=key=CFG_PARADIGM_LINK_ENABLE,value=Yes,group=General, PCFG_VARIABLE_BOLUS_ENABLED=key=PCFG_VARIABLE_BOLUS_ENABLED,value=Yes,group=Bolus, PCFG_ALARM_MODE=key=PCFG_ALARM_MODE,value=Silent,group=Sound, PCFG_AUDIO_BOLUS_ENABLED=key=PCFG_AUDIO_BOLUS_ENABLED,value=Yes,group=Bolus, PCFG_INSULIN_CONCENTRATION=key=PCFG_INSULIN_CONCENTRATION,value=50,group=Insulin, PCFG_MAX_BASAL=key=PCFG_MAX_BASAL,value=4.5,group=Basal, CFG_MM_RF_ENABLED=key=CFG_MM_RF_ENABLED,value=Yes,group=General, PCFG_INSULIN_ACTION_TYPE=key=PCFG_INSULIN_ACTION_TYPE,value=Curve: 5,group=Insulin, PCFG_MM_SRESERVOIR_WARNING_POINT=key=PCFG_MM_SRESERVOIR_WARNING_POINT,value=20,group=Other, PCFG_AUTOOFF_TIMEOUT=key=PCFG_AUTOOFF_TIMEOUT,value=0,group=General, PCFG_BASAL_PROFILES_ENABLED=key=PCFG_BASAL_PROFILES_ENABLED,value=No,group=Basal, CFG_MM_BLOCK_ENABLED=key=CFG_MM_BLOCK_ENABLED,value=No,group=General, PCFG_TEMP_BASAL_TYPE=key=PCFG_TEMP_BASAL_TYPE,value=Units,group=Basal, CFG_MM_KEYPAD_LOCKED=key=CFG_MM_KEYPAD_LOCKED,value=No,group=Other, PCFG_MAX_BOLUS=key=PCFG_MAX_BOLUS,value=12.5,group=Bolus, PCFG_BOLUS_SCROLL_STEP_SIZE=key=PCFG_BOLUS_SCROLL_STEP_SIZE,value=50,group=Bolus, PCFG_OTHER_DEVICE_PAIRED_STATE=key=PCFG_OTHER_DEVICE_PAIRED_STATE,value=Yes,group=Other, PCFG_AUDIO_BOLUS_STEP_SIZE=key=PCFG_AUDIO_BOLUS_STEP_SIZE,value=0.5,group=Bolus, PCFG_MM_RESERVOIR_WARNING_TYPE_TIME=key=PCFG_MM_RESERVOIR_WARNING_TYPE_TIME,value=PCFG_MM_RESERVOIR_WARNING_TYPE_UNITS,group=Other, PCFG_OTHER_DEVICE_ENABLE=key=PCFG_OTHER_DEVICE_ENABLE,value=Yes,group=Other, CFG_BASE_CLOCK_MODE=key=CFG_BASE_CLOCK_MODE,value=24h,group=General}.
08-04 08:04:09.559 4268-4268/com.gxwtech.roundtrip2 D/ShowAAPS2Activity: [main] DEBUG [com.gxwtech.roundtrip2.ShowAAPS2Activity:361]: Settings on front: {PCFG_CAPTURE_EVENT_ENABLE=key=PCFG_CAPTURE_EVENT_ENABLE,value=No,group=Other, CFG_PARADIGM_LINK_ENABLE=key=CFG_PARADIGM_LINK_ENABLE,value=Yes,group=General, PCFG_VARIABLE_BOLUS_ENABLED=key=PCFG_VARIABLE_BOLUS_ENABLED,value=Yes,group=Bolus, PCFG_ALARM_MODE=key=PCFG_ALARM_MODE,value=Silent,group=Sound, PCFG_AUDIO_BOLUS_ENABLED=key=PCFG_AUDIO_BOLUS_ENABLED,value=Yes,group=Bolus, PCFG_INSULIN_CONCENTRATION=key=PCFG_INSULIN_CONCENTRATION,value=50,group=Insulin, PCFG_MAX_BASAL=key=PCFG_MAX_BASAL,val [... cut ...] y=PCFG_BOLUS_SCROLL_STEP_SIZE,value=50,group=Bolus, PCFG_OTHER_DEVICE_PAIRED_STATE=key=PCFG_OTHER_DEVICE_PAIRED_STATE,value=Yes,group=Other, PCFG_AUDIO_BOLUS_STEP_SIZE=key=PCFG_AUDIO_BOLUS_STEP_SIZE,value=0.5,group=Bolus, PCFG_MM_RESERVOIR_WARNING_TYPE_TIME=key=PCFG_MM_RESERVOIR_WARNING_TYPE_TIME,value=PCFG_MM_RESERVOIR_WARNING_TYPE_UNITS,group=Other, PCFG_OTHER_DEVICE_ENABLE=key=PCFG_OTHER_DEVICE_ENABLE,value=Yes,group=Other, CFG_BASE_CLOCK_MODE=key=CFG_BASE_CLOCK_MODE,value=24h,group=General}`