andyrozman/RileyLinkAAPS #88: Rileylink BLE Scan does not find any device

Lytrix, 2018-08-15:
When scanning for a RL1, no devices are found or errors are showing. Had devices showing up once  and had the rileylink working, but after reinstalling I cannot retrieve a list of devices. This could be something: `D/BluetoothLeScanner( 7236): could not find callback wrapper`

```
V/ActivityThread( 7236): updateVisibility : ActivityRecord{348af885 token=android.os.BinderProxy@b7cdfa0 {com.gxwtech.roundtrip2/com.gxwtech.roundtrip2.RileyLinkScan}} show : false
V/ActivityThread( 7236): updateVisibility : ActivityRecord{348af885 token=android.os.BinderProxy@b7cdfa0 {com.gxwtech.roundtrip2/com.gxwtech.roundtrip2.RileyLinkScan}} show : true
V/ActivityThread( 7236): updateVisibility : ActivityRecord{348af885 token=android.os.BinderProxy@b7cdfa0 {com.gxwtech.roundtrip2/com.gxwtech.roundtrip2.RileyLinkScan}} show : true
D/BluetoothLeScanner( 7236): could not find callback wrapper
D/RileyLinkScan( 7236): scanLeDevice: Scanning Stop
V/ActivityThread( 7236): updateVisibility : ActivityRecord{348af885 token=android.os.BinderProxy@b7cdfa0 {com.gxwtech.roundtrip2/com.gxwtech.roundtrip2.RileyLinkScan}} show : true
D/BluetoothLeScanner( 7236): could not find callback wrapper
D/RileyLinkScan( 7236): scanLeDevice: Scanning Stop
I/Timeline( 7236): Timeline: Activity_idle id: android.os.BinderProxy@b7cdfa0 time:20991681
D/ViewRootImpl( 7236): ViewPostImeInputStage ACTION_DOWN
D/BluetoothLeScanner( 7236): onClientRegistered() - status=0 clientIf=6
D/RileyLinkScan( 7236): scanLeDevice: Scanning Start
D/RileyLinkScan( 7236): scanLeDevice: Scanning Stop
I/Timeline( 7236): Timeline: Activity_idle id: android.os.BinderProxy@b7cdfa0 time:20994271
D/BluetoothLeScanner( 7236): could not find callback wrapper
D/RileyLinkScan( 7236): scanLeDevice: Scanning Stop
D/SRIB_DCS( 7236): log_dcs ThreadedRenderer::initialize entered! 
I/Timeline( 7236): Timeline: Activity_idle id: android.os.BinderProxy@3e3ef879 time:21018411
D/SRIB_DCS( 7236): log_dcs ThreadedRenderer::initialize entered! 
I/Timeline( 7236): Timeline: Activity_idle id: android.os.BinderProxy@2fb99480 time:21019722
D/SRIB_DCS( 7236): log_dcs ThreadedRenderer::initialize entered! 
I/Timeline( 7236): Timeline: Activity_idle id: android.os.BinderProxy@e1b378d time:21020529
D/ViewRootImpl( 7236): ViewPostImeInputStage ACTION_DOWN
D/ViewRootImpl( 7236): ViewPostImeInputStage ACTION_DOWN
V/ActivityThread( 7236): updateVisibility : ActivityRecord{348af885 token=android.os.BinderProxy@b7cdfa0 {com.gxwtech.roundtrip2/com.gxwtech.roundtrip2.RileyLinkScan}} show : false
V/ActivityThread( 7236): updateVisibility : ActivityRecord{348af885 token=android.os.BinderProxy@b7cdfa0 {com.gxwtech.roundtrip2/com.gxwtech.roundtrip2.RileyLinkScan}} show : true
V/ActivityThread( 7236): updateVisibility : ActivityRecord{348af885 token=android.os.BinderProxy@b7cdfa0 {com.gxwtech.roundtrip2/com.gxwtech.roundtrip2.RileyLinkScan}} show : true
D/BluetoothLeScanner( 7236): could not find callback wrapper
D/RileyLinkScan( 7236): scanLeDevice: Scanning Stop
V/ActivityThread( 7236): updateVisibility : ActivityRecord{348af885 token=android.os.BinderProxy@b7cdfa0 {com.gxwtech.roundtrip2/com.gxwtech.roundtrip2.RileyLinkScan}} show : true
D/BluetoothLeScanner( 7236): could not find callback wrapper
D/RileyLinkScan( 7236): scanLeDevice: Scanning Stop
I/Timeline( 7236): Timeline: Activity_idle id: android.os.BinderProxy@b7cdfa0 time:20991681
D/ViewRootImpl( 7236): ViewPostImeInputStage ACTION_DOWN
D/BluetoothLeScanner( 7236): onClientRegistered() - status=0 clientIf=6
D/RileyLinkScan( 7236): scanLeDevice: Scanning Start
D/RileyLinkScan( 7236): scanLeDevice: Scanning Stop
I/Timeline( 7236): Timeline: Activity_idle id: android.os.BinderProxy@b7cdfa0 time:20994271
D/BluetoothLeScanner( 7236): could not find callback wrapper
D/RileyLinkScan( 7236): scanLeDevice: Scanning Stop
D/SRIB_DCS( 7236): log_dcs ThreadedRenderer::initialize entered! 
I/Timeline( 7236): Timeline: Activity_idle id: android.os.BinderProxy@3e3ef879 time:21018411
D/SRIB_DCS( 7236): log_dcs ThreadedRenderer::initialize entered! 
I/Timeline( 7236): Timeline: Activity_idle id: android.os.BinderProxy@2fb99480 time:21019722
D/SRIB_DCS( 7236): log_dcs ThreadedRenderer::initialize entered! 
I/Timeline( 7236): Timeline: Activity_idle id: android.os.BinderProxy@e1b378d time:21020529
D/ViewRootImpl( 7236): ViewPostImeInputStage ACTION_DOWN
D/ViewRootImpl( 7236): ViewPostImeInputStage ACTION_DOWN
```

Lytrix, 2018-08-15:
Ok,uncommented this line, and it's showing up after 2 seconds: https://github.com/andyrozman/RileyLinkAAPS/blob/dev_medtronic/app/src/main/java/com/gxwtech/roundtrip2/RileyLinkScan.java#L219

Lytrix, 2018-08-15:
And found this about the wrapper missing:
https://stackoverflow.com/questions/40862767/bluetoothlescanner-could-not-find-callback-wrapper#42821272

Lytrix, 2018-08-15:
This is triggering a stop scan action, when I comment this, it works as expected: https://github.com/andyrozman/RileyLinkAAPS/blob/dev_medtronic/app/src/main/java/com/gxwtech/roundtrip2/RileyLinkScan.java#L119

andyrozman, 2018-08-17:
I will have to take a lok at that and refactor it... onCreate shouldn't start the scanning, and onPause and onResume, shouldn't even do anything here... I will probably need your help to test this after its done, but this won't be soon...

andyrozman, 2018-09-09:
This is fixed, but there are several issues with this code so more detailed refactoring needs to be made (it works now, but it should be made much better). New ticket was opened #90 , closing this one.

closed 2018-09-09 by andyrozman