andyrozman/RileyLinkAAPS #101: AAPS: Give option to use RileyLink-internal 4b6b encoding

andyrozman, 2018-12-15:
New version of RileyLink (2.x) has internal encoding 4b6b (communication with Medtronic Pump). Most of software (AAPS and Loop) currently don't use it, so it was not really tested yet. 

Using internal decoding would be beneficial, since there would be less communication problems that way. 

We will give user option to select to use this RileyLinkInternal decoding (user options). So if user selects this, and has correct version of RileyLink, then this encoding will be enabled.

Bo7amada, 2018-12-16:
Hi Andy, sorry to say this here, didn't know where else to reach out.

I'm highly interested in this project (AAPS Medtronic pump support) and would love to contribute with all I could.

I have 3 loopable pumps for testing (522WW, 722NA, & 723NA), and an RL version 1.0, I also did use both Loop and OpenAPS before so I have that experience and base knowledge.

I can't code though, but I can compile builds and do testing, so plz let me know where to start and how I can help.

andyrozman, 2018-12-16:
Please join us in RileyLinkAAPS/Lobby room in gitter and we will get you running...

closed 2019-05-31 by andyrozman