andyrozman/RileyLinkAAPS #106: Add support for GNARL

andyrozman, 2019-01-30:
Add support for GNARL

GNARL is alternative to RileyLink device for connecting to Medtronic... Its an idea some of guys are toying about to use low cost device TTGO V2.0 instead of original RL.

https://www.aliexpress.com/item/TTGO-LORA32-V2-0-433-868-915Mhz-ESP32-LoRa-OLED-0-96-Inch-SD-Card-Blue/32846302183.html?spm=a2g0s.12269583.0.0.57004a16iDnlFr

winemug, 2019-01-31:
@andyrozman are there any more details to this idea? I have done my fair share of investigation on radios and ics for omnipod and would like to contribute.

andyrozman, 2019-01-31:
If you follow talks in hardware-dev, Eric is the main person behind that... This link should put you there: https://gitter.im/openaps/hardware-dev?at=5c4920be83189945243b30a0

His plan is to use different hardware with same firmware as RileyLink so that device would be totally exchangable without any changes.

As you can see this module comes in two varieties 433 and 866/900 Mhz, so making something similar for Omnipod should also work, and device has external antenna, which means we could get little bit more distance out of it.

closed 2019-06-13 by andyrozman