andyrozman/RileyLinkAAPS #107: Add support for Superbolus

peterlynton, 2019-02-13:
On medtronic_andy: when using Superbolus a message appears: 'Operation is not supported by pump and/or driver'.  Pump will continue with calculated bolus but will not set a TBR of 0% for two hours as it should.  I'm unsure if this can be implemented in the pump driver?

libxmike, 2019-02-14:
are you using 512/712 ? they don't support (e)SMB

andyrozman, 2019-02-14:
It should be supported. On any pump. This is not OpenAPS. Any Pump should support SMB. Can you please see what is in the logs? Some command was called that was not implemented.

peterlynton, 2019-02-14:
@andyrozman @libxmike To be clear:
[AndroidAPS.log](https://github.com/andyrozman/RileyLinkAAPS/files/2867145/AndroidAPS.log)
 This is for a Superbolus, not SMB - SMB are running fine on both 722 and 723.

Log attached, wizard used to calculate Superbolus at 22:54:07.379, and failing in setting 0% basal for 120min at 22:54:10.042.

andyrozman, 2019-02-20:
As we discovered this is AAPS core bug, so I am closing this one and Bug on AAPS was opened to fix this (they send wrong type of TBR to pump).

closed 2019-02-20 by andyrozman