andyrozman/RileyLinkAAPS #118: Need RileyLinks?

jlucasvt, 2019-05-01:
Devs,  

Please let me know if you need RileyLinks to assist with Development/Testing. 
email request to: jlucas@getrileylink.org

jroth-haj, 2019-05-01:
All testers have already bought one, without which we could not test. ;-)

jlucasvt, 2019-05-01:
Sorry I would have donated them to you if I had known you were doing dev.    Let me know if there's anything else you need from me.  Awesome to see Medtronic and Omnipod options heading to AAPS.

jroth-haj, 2019-05-01:
Andy ist doing a great job.
Currently I have once per week the issue that the Rileylink hangs off. Even if I stopped AAPS or reboot my phone, the Rileylink green LED still is on. Do you have an idea to solve this?
If so, please support Andy with your knowledge of Rileylink. That would be great!

After MDT is finished, he wants to work on omnipod.

winemug, 2019-05-02:
If you want to donate one to OmniPy/OmniCore project, I certainly wouldn't
say no.

Best,
Baris

jlucasvt, 2019-05-02:
> If you want to donate one to OmniPy/OmniCore project, I certainly wouldn't say no. Best, Baris
> […](#)

Send me an email jlucas@getrileylink.org with Name and Shipping Address and I'll send one of each out when we get them in from manufacture.