andyrozman/RileyLinkAAPS #126: Is it possible to integrate battery status (% or value) to MDT Tab and NS Pill ?

Alamo04, 2019-05-19:


andyrozman, 2019-05-20:
I changed title from this: 
Is it possible to integrate battery status (% or value) to MDT Tab and NS Pill ?

ToreBj, 2019-05-20:
Just in case you haven't read it:
https://loopkit.github.io/loopdocs/operation/features/battery/#pump-battery
Gives a rather good overview regarding alkaline/lithium battery differences. I guess Mysentry mode is not implemented(x23/x54 only) so using battery voltage covers all pumps.

ToreBj, 2019-05-20:
Some more battery info/tests:
https://bionicwookiee.com/2017/10/24/aaa-batteries-for-insulin-pumps/
https://bionicwookiee.com/2017/11/12/aaa-batteries-insulin-pumps-part-2/

Interesting table showing different batteries and how long they last:
[image]

ToreBj, 2019-05-23:
As a first I'm perfectly happy with the voltage only on the MDT tab.  The percentage calculation is difficult and not very accurate anyway. If the voltage is also sent to NS one can set an alarm limit for the low voltage there, unless there is already something built in to AAPS that can handle it.

stefanhubmer, 2019-06-18:
Maybe its possible to limit the voltage to 100%?
Today I got a percentage of 123% (1,64V)

Just a minor issue, so no need for a own ticket.

andyrozman, 2019-06-18:
What kind of battery do you use? How could you have 1.64V?

stefanhubmer, 2019-06-18:
I've got no idea. Using Energizer Ultimate Lithium. 
Just refreshed, now it's 103% (1,59V).

closed 2019-07-13 by andyrozman