andyrozman/RileyLinkAAPS #131: Error getting History - Crash APP

caiobarreto, 2019-06-04:
When the App is getting history from medtronic 715, when the process finished the App crash, I did a very quick debug and appear my pump sometimes return a wrong date, I ran a lot of times to create some bugs on Firebase, I think you will be able to see the error.

andyrozman, 2019-06-05:
Problem with this issue is that even looking at Logs I can't find the real error... I had 2 people with same problem and so far I couldn't reproduce, but I will look into it...

andyrozman, 2019-06-05:
Duplicate of #127

andyrozman, 2019-06-08:
It seems that problem is with Splitter from Guava, at least that is the problem on Android 6. I changed Splitter with my own implementation.

andyrozman, 2019-06-13:
Spliter replaced with my own implementation.

closed 2019-06-13 by andyrozman