andyrozman/RileyLinkAAPS #132: Setup wizard (pump config) Max Basal config UI tweaks

rustymonkey, 2019-06-05:
When configuring Max Basal, the default was something like 235 which made me think that perhaps it's asking for total basal/day. But I'm guessing that it's probably actually asking for Max Basal U/hr. So I would suggest:
1) Indicate units i.e. 'Max Basal on Pump (U/hr)'
related: 'Max Bolus on Pump (U)'
2) Set a default that is more realistic

p.s. you might also want to consider not relying on the numeric keyboard for data entry on this screen (at least for some devices, since the numeric keyboard on some Samsung devices is missing the decimal button).

andyrozman, 2019-06-05:
Setting needs to be a whole number, so no decimal points. But I will change description...

rustymonkey, 2019-06-05:
fyi the Max Basal setting on the pump is in the format of x.yz U/hr (i.e. not a whole number).

andyrozman, 2019-06-06:
Default was set to max (35)

rustymonkey, 2019-06-06:
Yes default Max Basal is 35 (not 235 as I originally wrote). My point (a very minor one) was that if average basals are e.g. 1u/hr and recommended Max Basal is e.g. 4x average basal, then a default of 4 might be more appropriate. Regardless, indicating the units will solve my confusion.

Re. whole numbers: What I meant is that on the pump, Max Basal is set to e.g. 4.3 (not a whole number). So you probably wouldn't want to restrict Max Basal to an integer.

closed 2019-06-13 by andyrozman