andyrozman/RileyLinkAAPS #135: Bolusing via calculator, get stuck at "waiting for pump"

stefanhubmer, 2019-06-12:
Using MDT_0.9 and Sony XZ1 Compact (Android 9)

When bolusing via calculator it sometimes get stuck at the "waiting for pump" popup. The "stop" button is not responding. Workaround: Close app and open it back again.
Pump is bolusing the correct amount.

As far as I can tell, the loop stops until you restart the app. Will update if I have more info

[image]

andyrozman, 2019-06-12:
What is for you stuck? There is certain amount of time (in s) that we wait before issuing the bolus. But stop should work...

andyrozman, 2019-06-12:
Process goes like this:
1. Trying to contact pump (this can be repeated up to 3 times, with 30 s of timeout each time
2. When pump is contacted we wait specific time (5/10/15 s, depending on configuration)
3. Bolus is issued.

Before step 3, cancel should be possible...

stefanhubmer, 2019-06-12:
Happened two times before and I'm using your MDT branch since five days.
Calculated, bolused, bolus given, stuck. Waited about 5 minutes (screen stayed as seen on screenshot), then killed the app.
After step 3, the popup didn't disappeared.

andyrozman, 2019-06-12:
It works good for everybody else... What kind of phone do you have?

stefanhubmer, 2019-06-12:
Sony G8441 XZ1 Compact (Android 9.0)
I'll take care of it and watch if it happens again in the next days. atm I can't give more infos

andyrozman, 2019-06-25:
@stefanhubmer Do you still get this problems with never releases 0.10 (stable) or 0.11.1 (latest working dev)

stefanhubmer, 2019-06-27:
@andyrozman Was using 0.10 about 2 days and got the same problem once. Unfortunately I'm on holiday and not using the MDT Branch atm, I'm going to start using back again at monday (1.7.) evening. If possible you can keep the ticket in backlog.

andyrozman, 2019-06-27:
Sure... When you come please upgrade to latest version... If last release performs, I might make release over weekend... (0.11)...

stefanhubmer, 2019-07-16:
Short update: unfortunately had no time to test, started yesterday with R0.11
Everythings smooth, I'm going to use it for two more days and if the bug won't appear again, I'll close the ticket.

andyrozman, 2019-07-18:
Please try with medtronic_andy_merge branch... that is the latest version... It should show as 2.4-dev...

stefanhubmer, 2019-07-21:
No issues with R0.11 (tested 2 days) and no issues with 2.4-dev (tested 3 days).
Used calculator 3-4 times every day.
Will close the ticket now

closed 2019-07-21 by stefanhubmer