andyrozman/RileyLinkAAPS #154: Pump status Suspend Issue

caiobarreto, 2019-07-08:
Showing pump status suspends, but the pump is not.

ElyeM, 2019-07-08:
Could be you are out of insulin? That's when I see suspended.

caiobarreto, 2019-07-08:
I have insulin, actually, I change the reservoir right now. But still showing "pump suspended"

[image]
[image]

andyrozman, 2019-07-08:
Log please. If you change reservoir it won't automatically go to "not suspended". It can take up to 5 minutes that state changes (when history is retrieved (every 5 minutes). 

If you want to force state refresh press "Refresh" button in Medtronic tab (you should do that after you change reservoir, battery anyways)...

Please supply me with log so that I can check if everything is ok... There might be some situation that would be incorrrectly setting "is pump suspended".

Theretically after 5 minutes your AAPS should be showing Pump running...

caiobarreto, 2019-07-08:
The status stills the same. Follow the log.
Actually, this situation happened more than once. After one day and after changed pump's battery the status updated.

[AndroidAPS_LOG_1562605246561.log.zip](https://github.com/andyrozman/RileyLinkAAPS/files/3369434/AndroidAPS_LOG_1562605246561.log.zip)

andyrozman, 2019-07-13:
You are using very old version medtronic-0.10.4... but as far as I can see the problem is in your history... 

13:57:08.437 [Thread-117] D/PUMP: [MedtronicHistoryData.showLogs():109]:
                [
  {
    "entryType": "PumpSuspend",
    "DT": "04.00.2030 00:30:02",
    "atechDateTime": 20300004003002
  },
  {
    "entryType": "PumpSuspend",
    "DT": "20.00.2026 00:30:18",
    "atechDateTime": 20260020003018
  },
  {
    "entryType": "TempBasalCombined",
    "DT": "08.07.2019 11:57:34",
    "atechDateTime": 20190708115734,
    "decodedData": {
      "Object": {
        "durationMinutes": 0,
        "insulinRate": 0.0,
        "isPercent": false
      }
    }
  },

You have this 2 entries in history one from 2030 and one from 2025, which always come out on top (all the history is writen after them... 

You will need to reset pump (to delete history) and then work from that... AND please install new version as soon as possible...

Closing this since this is not a bug, just local Pump issue.

closed 2019-07-13 by andyrozman