andyrozman/RileyLinkAAPS #158: Medronic 723: Basal profiles are note enabled on pump

rkresha, 2019-09-23:
I'm getting the following 'urgent' alert the first communication happens between AAPS and Rileylink. I have 3 basal settings enabled on the pump so I'm not sure what it is complaining about. Please advise and let me know what I can do to assist. I'm currently using the dev branch from the main AAPS github repo.

-Rob

jroth-haj, 2019-09-23:
AAPS ist only using the first basal profile.
Select the first profile!
Then so a Profile Switch in AAPS.
It will transfer your current selected profile to the pump.

rkresha, 2019-09-24:
@jroth-haj I'm using the 'simple profile' profile if that makes a difference, however I'm not sure I follow what you suggesting I should do, could you please reword it?

rkresha, 2019-09-24:
Switched over to local profile and did an activate profile and then a switch profile. I hit  'refresh' and get the same urgent alert

rkresha, 2019-12-28:
Error message adjusted to be more intuitive for what the fix is.

closed 2019-12-28 by rkresha