andyrozman/RileyLinkAAPS #159: Duplicate treatments

ElyeM, 2019-09-27:
I'm getting a lot of duplicate treatments lately.

I opened the calculater wizard, entered 20 carbs, checked super bolus, OK.

Now when I go to treatments I see duplicate entries, one Minute apart. One is a meal entry the other is a correction entry. 

AAPS Version: RC2
Pump model: 723
Time of event: 4:03/4:04PM

See screenshots 

[image]
[image]

ElyeM, 2019-09-29:
Another duplicate, now one comes in as SMB and the next as corr, also 1 minute apart. 

2:39/2:40

[image]
[image]

peterlynton, 2019-10-09:
I'm also getting duplicate entries.  I'm attaching screenshots and log files from one event, but have more if required.

AAPS Version: RC2
Pump model: 722
Time of event: 09:12/09:14
[image]
[image]
[AndroidAPS._2019-10-02_00-00-00_.1.zip](https://github.com/andyrozman/RileyLinkAAPS/files/3708383/AndroidAPS._2019-10-02_00-00-00_.1.zip)
[AndroidAPS._2019-10-02_00-00-00_.0.zip](https://github.com/andyrozman/RileyLinkAAPS/files/3708384/AndroidAPS._2019-10-02_00-00-00_.0.zip)

andyrozman, 2019-10-09:
Did you get any warning that time could not be set?